meerklass/museek #70: scan and track data

mariogrs, 2023-11-09:
It's weird that the printing the object with the scan data, prints the log with all the data:
scan_data = context_loader.get_result(location=ResultEnum.SCAN_DATA)
print(scan_data)

Same goes with track data.

Also, track data seems to include all the time dumps, not just the track ones.

amadeuswi, 2023-11-13:
When we run `print(data)` or `print(scan_data)` the output is always the same. This is because at the moment both share the same `print()` function. We could implement `print` to reflect the selection of data. Until now I always used `data.timestamps` to check the timestamp selection.

mariogrs, 2023-11-14:
Not critical but I I suggest to at least print a warning because people might think the scan data includes everything otherwise.