meerklass/museek #74: possible problem with sun check output

mariogrs, 2023-11-19:
I tried the new sanity check with sun on uhf block 1666630286 and got a weird value for the time difference:

check closeness to sunset/sunrise: 
No Good, the time difference between start/end time and sunset/sunrise is 
0.4684/-519.2994 minutes.

wkhu-astro, 2023-11-20:
The positive value means the start/end time is after sunset/sunrise, the negative value means the start/end time is before sunset/sunrise

mariogrs, 2023-11-20:
but surely not 519 minutes?

wkhu-astro, 2023-11-20:
This observation was observed from 2022-10-24 18:51:53.278 SAST to 2022-10-24 21:05:36.905 SAST,   the Sunset time: 2022-10-24 18:51:26 SAST, Sunrise time: 2022-10-25 05:44:54 SAST,  Considering the observation is usually carried out at night, the used sunset time is the nearest sunset time before observation start, the used sunrise time is the nearest sunrise time after observation end.

closed 2023-11-22 by mariogrs