meerklass/museek #84: Running sanity check with token not currently possible

piyanatk, 2024-07-16:
Running sanity check with the rdb token is currently not possible. `katdal` will produce an error, suggesting that there is a network connection issue to the SARAO archive. This might be due to SARAO office being moved, and thus changes in their network connectivity.
```python
 File "/users/piyanat/miniconda3/envs/meerklass/lib/python3.10/site-packages/katdal/datasources.py", line 467, in from_url
    raise DataSourceNotFound(str(e)) from e
katdal.datasources.DataSourceNotFound: HTTPSConnectionPool(host='archive-gw-1.kat.ac.za', port=443): Max retries exceeded with url: /1720980926/1720980926_sdp_l0.full.rdb (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x7f406d2b1d80>: Failed to resolve 'archive-gw-1.kat.ac.za' ([Errno -2] Name or service not known)"))
```

Our current workaround is to first download the data.

Boom will submit a ticket to the helpdesk and follow up.

mariogrs, 2024-07-23:
I've run it and it worked. But sometimes it doesn't work. Probably because of a network break. Ideally we should make the code robust to this. A quick fix would be to use #SBATCH --requeue

piyanatk, 2024-07-24:
Yes, it works again now. I agree with adding `--requeue`. Should we put the bash script in the repo? Since you can actually override plugin parameters by passing to `museek`, e.g. `museek --InPlugin-block-name ${block_name} --InPlugin-token ${token} museek.config.sanity_check`, we can probably keep the plugin config as a static template, and make an sbash script that take user input options that override the token, and etc.

mariogrs, 2024-07-24:
I was actually not aware of that. Yes, that would make things much simpler. Can you please upload the batch script then? I'm not quite sure what are the minimum resources required yet for the sanity check. Timewise, I assume after 30 min we should kill it anyway.

piyanatk, 2024-07-24:
Yes, I will do so. I will look into the resources required. If you save the vis, the peak memory usageis about 128GB, which is pretty bad, but that might change if we don't need to save the vis.

piyanatk, 2024-07-25:
Close as the issue appeared to be occasional network issue from SARAO side.

closed 2024-07-25 by piyanatk