meerklass/museek #85: Sanity check - dish elevation

mariogrs, 2024-07-24:
Confirm if the elevation check for the dishes (if they keep the correct elevation) takes into account what we call stragglers - dishes that have been identified by the office as not moving correctly. I think this might affect the calculation if it's not correct.

piyanatk, 2024-07-28:
@mariogrs I am still trying to understand the problem. The stragglers are dishes that got stuck during slew, so they ended up missing some scans, correct? Are these incidents not accounted for in all the elevation plots?

mariogrs, 2024-07-29:
Actually, I think it's better for Wenkai to look at this since he wrote the elevation checks. Let me assign to him.

wkhu-astro, 2024-08-03:
The elevation check in the sanity check will be performed for each antenna individually, including those that are stragglers.

mariogrs, 2024-08-05:
Then I think this is fine. We can close the issue.

closed 2024-08-05 by mariogrs