meerklass/museek #87: Point source list

mariogrs, 2024-07-31:
Create a text based file with a list of point sources on the sky above a certain flux. Ideally we should have a spectral index model or the fluxes at different frequencies. Maybe use a 5 Jy cut at 1.4 GHz.  At 800 MHz this would give 7 Jy assuming a spectral index of alpha=-0.7. A 10 Jy source at a given frequency will generate a temperature of about 0.5K at that frequency. This is about the same level of fluctuations as the galactic synchrotron. So it shouldn't be picked up by the flagger, although it might affect the map rms if there are many of those. Of course, if the spectral index is more steep, a 5 Jy cut at 1.4 GHz might still allow for strong sources at the UHF frequencies (550 - 1050 MHz).

mariogrs, 2024-08-05:
@physicistsam ?

mariogrs, 2024-08-05:
Reopening and waiting for @physicistsam to confirm.

physicistsam, 2024-08-16:
I have all the possible notifications switched on, but still not getting anything from this. Apologies. I shall work on it now.

mariogrs, 2024-08-17:
Maybe check spam? Other people are getting notifications.

physicistsam, 2024-08-19:
Here is the version 0 of the catalogue. The txt file has four columns- RA, DEC, NVSS Flux at 1400 MHz, Spectral index.  
I will try to improve upon this. 
[5jy_scat-V0.txt](https://github.com/user-attachments/files/16658167/5jy_scat-V0.txt)

PS: I am able to get notifications now.

mariogrs, 2024-08-19:
Please check with Wenkai since he already produced a catalogue that he is
using. We should converge on the format. Let's discuss this here.

mariogrs, 2024-09-02:
can we agree on the catalogue format? txt format? what columns?
@physicistsam @wkhu-astro @spinemart @piyanatk ?

physicistsam, 2024-09-02:
@mariogrs, the txt file has four columns- RA, DEC, NVSS Flux at 1400 MHz, Spectral index. @wkhu-astro may have some input on how the module is reading the file.

spinemart, 2024-09-02:
My proposition was to make this more general and leave empty the info we do
not have atm:
Name, Ra, Dec, freq at which flux, flux at that freq, spectral index, flag
for the AO flagger.

piyanatk, 2024-09-03:
I second @spinemart, but we should make it clear whether the flux is integrated or peak value, and we might want to consider have more than one pair of (flux, frequency) if  we want interpolate between subbands rather than using spectral indices. Given that we are in U and L band, I supposed we do not really have source catalogues with subbands in UHF, so spectral index is probably the way to go?

mariogrs, 2024-09-04:
Let's go with: Name, Ra, Dec, Flux, Freq, spectral index (if available), Source info (maybe link - relevant to know if this is single dish or interferometer measured flux), flag for the AO flagger. I suggest to use a column or "|" to separate the columns. We can add more columns later with more fluxes.

mariogrs, 2024-09-17:
New format: Name, Ra, Dec, Flux, Freq, spectral index, Source info (maybe link - relevant to know if this is single dish or interferometer measured flux). Use "|" to separate the columns - I'm not sure if python takes more than one character as separator. Leave column empty if there is no info.

physicistsam, 2024-09-20:
@mariogrs and @wkhu-astro Please find the updated source catalog. It follows the format mentioned above. [5jy_scat-V0_new.txt](https://github.com/user-attachments/files/17074896/5jy_scat-V0_new.txt)

physicistsam, 2024-10-15:
This source catalog contains sources upto 5 Jy at 1.4 GHz. Better sky model is required in future.

physicistsam, 2024-10-23:
I am keeping a source catalog contains sources upto 1 Jy at 1.4 GHz. [1jy_scat-V0_new.txt](https://github.com/user-attachments/files/17490768/1jy_scat-V0_new.txt)

physicistsam, 2025-03-25:
Updated version of the catalog. I have added a cross-matched source from  [Parkes Southern Radio Source Catalog](https://heasarc.gsfc.nasa.gov/W3Browse/all/pkscat90.html). Now the columns are as follows.

Source Match ID, RA, DEC, NVSS_flux, Spectral Index between TGSS and NVSS, PKS Jname, PKS flux at 1410MHz, PKS flux at 635MHz, PKS flux at 408MHz.

Please note that all the sources are not matched with PKS catalog. Some of the PKS fluxes are not available. This source catalog contains sources with minimum 1 Jy flux at 1.4 GHz.

[1jy_scat-V1_new.txt](https://github.com/user-attachments/files/19457779/1jy_scat-V1_new.txt)

mariogrs, 2025-03-26:
Can you include a column for the 3C name and another for the PKS name? (if available). And add a line at the top with the meaning of each column, e.g.: # 3C ID # PKS ID # Flux # etc...

physicistsam, 2025-03-29:
@mariogrs thanks for the suggestions. I have updated the list with appropriate columns and the first row indicates the column names. 

#SOURCE_ID | RA | DEC | NVSS_FLUX | INT_SPEC_INDX | 3C_NAME | PKS_NAME | PKS_1410_FLUX | PKS_635_FLUX | PKS_408_FLUX | PKS_SPEC_INDX 

Atlease for the sources that hvae minimum three measurement of the fluxes at different frequencies, I have fitted a power law to estimate the single-dist spectral index and that is mentioned in the last colum. PKS catalog has flux information at nine frequencies. But all of them doesn't have a flux measurement. I have used only these bands ([80, 178, 408, 635, 1410, 2700] MHz) to estimate the spectral index. 

NVSS-TGSS catalog has 1617 number of sources below a declination of 30 degrees, with flux densities more than 1 Jy at 1400 MHz. PKS matchs with 1241 of them. Whereas, 188 3C sources are matched with the NVSS-TGSS catalogued sources. 

Below I am attaching two plots 1) NVSS-PKS flux comparison at 1400 MHz between the matched sources. This shows a quite tight relation. 2) A histogram of spectral index (SI) from NVSS-TGSS and PKS. Not all the PKS sources have a measured SI because of the three frequency measurement criterian. However the matched sources show a resonably idendical SI distribution.  

Updated catalog: 

[1jy_scat-V3_new.txt](https://github.com/user-attachments/files/19522502/1jy_scat-V3_new.txt)

[image]

[image]

physicistsam, 2025-04-08:
Catalog based on PKS can be used.

[1jy_scat-V3_single_dish.txt](https://github.com/user-attachments/files/19648081/1jy_scat-V3_single_dish.txt)

physicistsam, 2025-04-18:
Joint catalog. Updated.

[1jy_scat-V4_new.txt](https://github.com/user-attachments/files/19813462/1jy_scat-V4_new.txt)

mariogrs, 2025-04-18:
Some of the strong sources still seem to be missing from this catalogue. The catalogue needs to be completed above 10 Jy in order to be used. At least I cannot find all the ones listed below:

Calibrator | RA | Dec | flux_600MHz | flux_1000MHz
Cen A | 201.365063 | -43.019113 | 736.9690198 | 129.1676021
Pictor A | 79.957171 | -45.778828 | 124.5971065 | 85.22094494
Hercules A | 252.783945 | 4.992588 | 113.5016382 | 67.0062615
3C353 | 260.117326 | -0.979617 | 103.0685209 | 70.0348386
Hydra A | 139.52362 | -12.095543 | 93.45787351 | 59.16736675
3C273 | 187.277915 | 2.052388 | 50.63802008 | 45.28160486
3C161 | 96.792155 | -5.884772 | 33.97259213 | 23.62351271
PKS 1932-46 | 293.985649 | -46.344601 | 27.73298273 | 17.30560376

physicistsam, 2025-04-19:
I could find all the source in the catalog mentioned above except Cen A, the sources are placed in the same order below:

#SOURCE_ID | RA | DEC | NVSS_FLUX | INT_SPEC_INDX | 3C_NAME | PKS_JNAME | PKS_1410_FLUX | PKS_635_FLUX | PKS_408_FLUX | PKS_SPEC_INDX 
No-ID|79.934583|-45.777778|0.000000|0.000000|No-3C|PKS-J0519-4546|66.000000|0.000000|166.000000|-0.729482
564352|252.787116|4.991248|47.255966|-0.939120|3C[348]|PKS-J1651+0459|47.000000|0.000000|169.000000|-0.986481
1055679|260.125591|-0.979783|55.670414|-0.711213|3C[353]|PKS-J1720-0058|54.000000|97.000000|138.000000|-0.653160
726598|139.524312|-12.093676|42.992828|-0.876014|3C[218]|PKS-J0918-1205|43.500000|0.000000|132.000000|-0.832145
11347|187.276513|2.051155|55.339714|-0.322505|3C[273]|PKS-J1229+0203|42.000000|0.000000|55.100000|-0.333677
1113687|96.791588|-5.885657|19.157124|-0.663406|3C[161]|PKS-J0627-0553|18.500000|0.000000|44.700000|-0.630787
No-ID|293.985417|-46.344167|0.000000|0.000000|No-3C|PKS-J1935-4620|12.600000|0.000000|39.600000|-0.881485

physicistsam, 2025-04-19:
I investigated the issue with Cen A, it seems that the Parkes Radio Sources Catalogue (PKSCAT90) (Wright+ 1990) does not have a measurement of Cen A at 635 MHz or 1410 MHz (a screen shot is attached) and thats why it has been left out. Now I have incorporated that, now, any source with atleast one measurement with equivalent flux of 1 Jy at 1410 MHz with SI -0.7, is included. 
```
x=np.asarray([80, 178, 408, 635, 1410, 2700])
#1 Jy cut at 1410 MHz
flux_cut=np.asarray([(80/1410)**(-0.7), (178/1410)**(-0.7), (408/1410)**(-0.7), (635/1410)**(-0.7), 1, (2700/1410)**(-0.7)])
# 5Jy cut at 1410 MHz
flux_cut_5Jy = 5*flux_cut
```

I understand that its easier for the user if the A-team SOURCE_ID are updated with the names, I have done so by-hand, as the catalogs doesn't supply that.
```
#SOURCE_ID | RA | DEC | NVSS_FLUX | INT_SPEC_INDX | 3C_NAME | PKS_JNAME | PKS_1410_FLUX | PKS_635_FLUX | PKS_408_FLUX | PKS_SPEC_INDX

PIC-A|79.934583|-45.777778|0.000000|0.000000|No-3C|PKS-J0519-4546|66.000000|0.000000|166.000000|-0.729482

PUP-A|126.030833|-42.996111|0.000000|0.000000|No-3C|PKS-J0824-4259|18.000000|0.000000|198.000000|0.000000

VIR-A|187.705512|12.388655|142.316559|-0.550627|3C274|PKS-J1230+1223|220.000000|0.000000|519.000000|-0.713582

TAU-A|83.630784|22.018364|727.483215|0.699894|3C144|PKS-J0534+2200|0.000000|0.000000|0.000000|0.000000

CEN-A|201.383333|-43.066389|0.000000|0.000000|No-3C|PKS-J1325-4303|0.000000|0.000000|2746.000000|0.000000

FOR-A|50.651667|-37.239167|0.000000|0.000000|No-3C|PKS-J0322-3714|0.000000|0.000000|259.000000|0.000000
```
Please find both 1Jy and 5Jy catalogs below.

[5jy_scat-V5_new.txt](https://github.com/user-attachments/files/19819138/5jy_scat-V5_new.txt)
[1jy_scat-V5_new.txt](https://github.com/user-attachments/files/19819137/1jy_scat-V5_new.txt)

mariogrs, 2025-05-22:
I found at least one repeated source in the table. Could you remove any repeated entries?

TAU-A | 83.630784 | 22.018364 | 727.483215 | 0.699894 | 3C144 | PKS-J0534+2200 | 0 | 0 | 0 | 0 

103436 | 83.630773 | 22.018374 | 727.483215 | 0.695082 | 3C144 | PKS-J0534+2200 | 0 | 0 | 0 | 0

mariogrs, 2025-10-26:
and what is the definition of the spectral index?

physicistsam, 2025-10-28:
Thanks @mariogrs  for noticing this. I have fixed the issues mentioned. A-team source IDs are put by hand, I have included them at the srart of the file for easier access. Please use this updated version.

[1jy_scat-V8_new.txt](https://github.com/user-attachments/files/23193203/1jy_scat-V8_new.txt)