meerklass/museek #91: Offset correction

mariogrs, 2024-08-16:


mariogrs, 2024-08-21:
Let's please discuss things here @wkhu-astro @spinemart . 
The idea to use the average maps to calibrate the offset instead of picking a reference map is that the scatter due to noise will be smaller. But I think we need to make sure the average in each pixel is done over the same scans. We don't need pixels with all the scans/blocks. But we need to make sure they all use the same scans for the average.

wkhu-astro, 2024-08-24:
I did the offset correction for 2023 data in desi1 :block_name_list =  ['1684087370', '1682448988', '1679605292', '1676657789', '1676313206', '1675816512', '1675643846', '1675623808', '1675210948', '1675021905']. Below are some results. 
This is the overlap region of these blocks, I will use these pixels to calculate the offset.
[image]
For the frequency, I select the frequency points that are not masked in any blocks, the used frequency is shown below:
[image]
Then, I take the median for these frequencies and then calculate the offset. 
The weighted average of maps versus the synch model (both maps and model are at the the median of selected frequencies) :
[image]
The weighted average of maps versus the each individual map:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Apply the offset (& slope) correction to each block and redo the weighted average, below is the comparison between combined maps without and with correction.
[image]
[image]
[image]
[image]

I also fit the offset (& slope ) at each selected frequency point and plot the offset (& slope ) as a function of frequency: 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mariogrs, 2024-08-29:
what is the difference between the red and black lines in the T_aver versus T_pix plots above @wkhu-astro ?

wkhu-astro, 2024-08-29:
> what is the difference between the red and black lines in the T_aver versus T_pix plots above @wkhu-astro ?

For the T_aver vs. T_model, the red-dashed line refers to f(x) = x. For the T_aver vs. T_pixel, the red-dashed line is the fitting    results, the black-dashed line is f(x) = x.

wkhu-astro, 2024-08-30:
The index vs. T_pixel for each block:  
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

wkhu-astro, 2024-10-17:
I split the data used for offset correction into two sub set (sub1 and sub2): 
[image]
[image]
I run the offset correction for the two sub data, the spectrum after offset correction:
[image]
[image]

wkhu-astro, 2024-10-17:
"check if final pixel versus frequency still shows the "square drops" if you remove the block which was flagged in those regions" 

I remove one block and then do the offset correction, the results with all blocks and one block removed:
[image]
[image]
[image]

wkhu-astro, 2025-04-29:
A self-calibration approach is added after the offset correction described above, there is still a few small offsets, but the results are much better.