meerklass/museek #94: sanity check should plot scan tracks of more than one antennas

piyanatk, 2024-09-05:
At the moment, the sanity check plug-in takes a `reference_receiver_index` parameter, which is basically the antenna/dish number, e.g. `reference_receiver_index=0` refers to `m001`. Only the scan tracks of this antenna is plotted, but sometimes this antenna turns out to be a strangler, or something happened, showing an incomplete scan track. Below is an example from `1720370484`
[image]
[image]

Thus, the sanity check should ideally plot scan tracks of more than one antennas for this reason.

mariogrs, 2024-09-05:
I agree. Perhaps we could plot the median instead? If the majority of
dishes are behaving as expected the median would give the correct path.

piyanatk, 2024-09-05:
@mariogrs I think plotting the median is a good idea for the first plot. For the zoom in, we don't necessarily need to see the shape of the HRS track right since it is already appear in the first plot. Instead of that, I think it will be more useful to plot the mean, median, and shaded std around the mean for the scan-only tracks as a function of time.