meerklass/museek #152: dec az issue with sanity checks

mariogrs, 2025-03-02:
museek/plugin/sanity_check_observation_plugin.py

Some of the blocks have completely crazy values for de min/max and az min/max from the sanity check output. The real ra and dec look fine.

wkhu-astro, 2025-03-20:
The reasons are that (1) there are outlier antennas (2) the pointing (all antennas) going far away from the expected scan region during scan.  

[image]
[image]
Figure1: example for reason (1), Left: outlier antenna which is not selected as straggler by SARAO, Right: good antenna.

[image]
Figure 2: example for reason 2, all antennas pointed wrong position at a certain time during scanning period.

I update the code for calculating the minimum of dec to like "`dec_min = np.min(np.median(data.declination.array[:, :, no_straggler_indexes], axis=-1))`",  this can avoid the influence from outlier antennas, but for the mispointing during one scan, we need to mask this to fix it.  I'm trying to do this.

mariogrs, 2025-03-20:
aren't all these maskings already implemented in the museek plugin?

wkhu-astro, 2025-03-21:
The SARAO mask can flag the situation in reason 2 (all antennas pointing wrong position), but for reason 1, outlier antennas may not be masked.  I have updated the sanity_check_plugin, and the mask is loaded, so we should run the sanity check after the data (flags) are downloaded.

piyanatk, 2025-04-30:
So I think @wkhu-astro addressed this in https://github.com/meerklass/museek/commit/0c825df59cef4e6d028da753cca4a11bc3849247, but it made the sanity check plugin load the full auto-correlation data. I revert @wkhu-astro's change in https://github.com/meerklass/museek/pull/164/commits/101633f45c26ab37a666ff64efdce7118e08d10e.

I think there are a few solutions to this issue that does not require loading the flags

1. Make the code "validate" the RA-Dec range, and auto switch to a different reference antenna if the range does not make sense. I think this should be doable by looking at skewness of the RA-Dec distribution or maybe use 1.5 x IQR as a threshold.
2. Add `ra_size` and `dec_size` parameters that users can pass. Then you can do,
```python
if dec_max > dec_mean + dec_size / 2:
    # raise warning and re-calculate ra, dec, az with a new reference antennas.
```

mariogrs, 2025-05-02:
I think the taking the median will take care of some outlier dishes. There is just some unnecessary repetition here:

dec_min = np.ma.min(np.ma.median(dec[:, :, no_straggler_indexes], axis=-1))
dec_max = np.ma.max(np.ma.median(dec[:, :, no_straggler_indexes], axis=-1))
dec_mean = np.ma.mean(np.ma.median(dec[:, :, no_straggler_indexes], axis=-1))
ra_min = np.ma.min(np.ma.median(ra[:, :, no_straggler_indexes], axis=-1))
ra_max = np.ma.max(np.ma.median(ra[:, :, no_straggler_indexes], axis=-1))
ra_mean = np.ma.mean(np.ma.median(ra[:, :, no_straggler_indexes], axis=-1))

For the time where ALL dishes do something weird, we could add an extra step to flag those times. For instance, in the log text we have:
Azimuth scan extent [10.8, -10.8]', '2024-02-23 18:05:58.199Z INFO     Scan target: scan_azel_with_nd_trigger, tags=azel, 81:03:57.6 41:43:28.4,

So we can extract the elevation and azimuth range. Any time with coordinates outside this can be flagged. Of course there is no reason for dishes to go outside this range besides wind stow, so we could simply use this range to calculate the ra,dec range.

piyanatk, 2025-09-02:
Fixed.