meerklass/museek #154: import matplotlib error

stevecunnington, 2025-03-18:
I'm trying to set up a museek environment from scratch on ilifu and getting a very simple matplotlib import error in the notebook. I've recreated the problem and pasting my full terminal output here (sorry it's so long). Any clever people (@wkhu-astro @kfornaz @piyanatk ) able to fix this in the GitHub repo? I assume everyone will encounter this who is trying to clone and run for the first time? Sorry if I've done something stupid...

```
(base) scunnington@compute-201:~$ git clone git@github.com:meerklass/museek.git
Cloning into 'museek'...
remote: Enumerating objects: 3879, done.
remote: Counting objects: 100% (830/830), done.
remote: Compressing objects: 100% (352/352), done.
remote: Total 3879 (delta 542), reused 485 (delta 478), pack-reused 3049 (from 2)
Receiving objects: 100% (3879/3879), 288.74 MiB | 7.69 MiB/s, done.
Resolving deltas: 100% (2645/2645), done.
(base) scunnington@compute-201:~$ git clone git@github.com:meerklass/ivory.git
Cloning into 'ivory'...
remote: Enumerating objects: 1048, done.
remote: Counting objects: 100% (136/136), done.
remote: Compressing objects: 100% (119/119), done.
remote: Total 1048 (delta 34), reused 48 (delta 17), pack-reused 912 (from 1)
Receiving objects: 100% (1048/1048), 191.34 KiB | 191.00 KiB/s, done.
Resolving deltas: 100% (725/725), done.
(base) scunnington@compute-201:~$ module load python/3.10.4
(base) scunnington@compute-201:~$ virtualenv ./environment/museek
created virtual environment CPython3.10.4.final.0-64 in 4774ms
  creator CPython3Posix(dest=/users/scunnington/environment/museek, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/users/scunnington/.local/share/virtualenv)
    added seed packages: Cython==3.0.12, MarkupSafe==3.0.2, PyJWT==2.10.1, PyYAML==6.0.2, alabaster==1.0.0, astropy==6.1.7, astropy_iers_data==0.2025.3.17.0.34.53, asttokens==3.0.0, async_timeout==5.0.1, attrs==25.3.0, babel==2.17.0, certifi==2025.1.31, charset_normalizer==3.4.1, cityhash==0.4.7, click==8.1.8, cloudpickle==3.1.1, comm==0.2.2, dask==2024.8.1, debugpy==1.8.13, decorator==5.2.1, docutils==0.21.2, ephem==4.2, exceptiongroup==1.2.2, executing==2.2.0, fsspec==2025.3.0, future==1.0.0, h5py==3.13.0, healpy==1.18.0, hiredis==3.1.0, idna==3.10, imagesize==1.4.1, importlib_metadata==8.6.1, iniconfig==2.0.0, ipykernel==6.29.5, ipyparallel==9.0.1, ipython==8.34.0, jedi==0.19.2, jinja2==3.1.6, joblib==1.4.2, jupyter_client==8.6.3, jupyter_core==5.7.2, katdal==0.23, katpoint==0.10.2, katsdptelstate==0.14, llvmlite==0.43.0, locket==1.0.0, matplotlib_inline==0.1.7, mock==5.2.0, msgpack==1.1.0, nest_asyncio==1.6.0, numba==0.60.0, numpy==1.23.3, packaging==24.2, parso==0.8.4, partd==1.4.2, pexpect==4.9.0, pip==25.0.1, platformdirs==4.3.6, pluggy==1.5.0, prompt_toolkit==3.0.50, psutil==7.0.0, ptyprocess==0.7.0, pure_eval==0.2.3, py==1.11.0, pyephem==9.99, pyerfa==2.0.1.5, pygments==2.19.1, pytest==7.1.3, python_dateutil==2.9.0.post0, python_lzf==0.2.6, pyzmq==26.3.0, rdbtools==0.1.15, redis==5.2.1, requests==2.32.3, scikit_learn==1.6.1, scipy==1.13.1, setuptools==75.8.2, six==1.17.0, snowballstemmer==2.2.0, sphinx==8.1.3, sphinxcontrib_applehelp==2.0.0, sphinxcontrib_devhelp==2.0.0, sphinxcontrib_htmlhelp==2.1.0, sphinxcontrib_jsmath==1.0.1, sphinxcontrib_qthelp==2.0.0, sphinxcontrib_serializinghtml==2.0.0, stack_data==0.6.3, threadpoolctl==3.6.0, tomli==2.2.1, toolz==1.0.0, tornado==6.4.2, tqdm==4.67.1, traitlets==5.14.3, typing_extensions==4.12.2, urllib3==2.3.0, wcwidth==0.2.13, wheel==0.45.1, zipp==3.21.0
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
(base) scunnington@compute-201:~$ source ./environment/museek/bin/activate
(museek) (base) scunnington@compute-201:~$ cd ivory
(museek) (base) scunnington@compute-201:~/ivory$ pip install --editable .
Obtaining file:///users/scunnington/ivory
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy~=1.23.3 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ivory==0.0.1) (1.23.3)
Requirement already satisfied: py~=1.11.0 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ivory==0.0.1) (1.11.0)
Requirement already satisfied: pytest~=7.1.3 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ivory==0.0.1) (7.1.3)
Collecting setuptools~=59.6.0 (from ivory==0.0.1)
  Downloading setuptools-59.6.0-py3-none-any.whl.metadata (5.0 kB)
Requirement already satisfied: joblib in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ivory==0.0.1) (1.4.2)
Requirement already satisfied: ipyparallel in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ivory==0.0.1) (9.0.1)
Requirement already satisfied: attrs>=19.2.0 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from pytest~=7.1.3->ivory==0.0.1) (25.3.0)
Requirement already satisfied: iniconfig in /users/scunnington/environment/museek/lib/python3.10/site-packages (from pytest~=7.1.3->ivory==0.0.1) (2.0.0)
Requirement already satisfied: packaging in /users/scunnington/environment/museek/lib/python3.10/site-packages (from pytest~=7.1.3->ivory==0.0.1) (24.2)
Requirement already satisfied: pluggy<2.0,>=0.12 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from pytest~=7.1.3->ivory==0.0.1) (1.5.0)
Requirement already satisfied: tomli>=1.0.0 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from pytest~=7.1.3->ivory==0.0.1) (2.2.1)
Requirement already satisfied: decorator in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipyparallel->ivory==0.0.1) (5.2.1)
Requirement already satisfied: ipykernel>=6.9.1 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipyparallel->ivory==0.0.1) (6.29.5)
Requirement already satisfied: ipython>=5 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipyparallel->ivory==0.0.1) (8.34.0)
Requirement already satisfied: jupyter-client>=7 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipyparallel->ivory==0.0.1) (8.6.3)
Requirement already satisfied: psutil in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipyparallel->ivory==0.0.1) (7.0.0)
Requirement already satisfied: python-dateutil>=2.1 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipyparallel->ivory==0.0.1) (2.9.0.post0)
Requirement already satisfied: pyzmq>=25 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipyparallel->ivory==0.0.1) (26.3.0)
Requirement already satisfied: tornado>=6.1 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipyparallel->ivory==0.0.1) (6.4.2)
Requirement already satisfied: tqdm in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipyparallel->ivory==0.0.1) (4.67.1)
Requirement already satisfied: traitlets>=5 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipyparallel->ivory==0.0.1) (5.14.3)
Requirement already satisfied: comm>=0.1.1 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipykernel>=6.9.1->ipyparallel->ivory==0.0.1) (0.2.2)
Requirement already satisfied: debugpy>=1.6.5 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipykernel>=6.9.1->ipyparallel->ivory==0.0.1) (1.8.13)
Requirement already satisfied: jupyter-core!=5.0.*,>=4.12 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipykernel>=6.9.1->ipyparallel->ivory==0.0.1) (5.7.2)
Requirement already satisfied: matplotlib-inline>=0.1 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipykernel>=6.9.1->ipyparallel->ivory==0.0.1) (0.1.7)
Requirement already satisfied: nest-asyncio in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipykernel>=6.9.1->ipyparallel->ivory==0.0.1) (1.6.0)
Requirement already satisfied: exceptiongroup in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipython>=5->ipyparallel->ivory==0.0.1) (1.2.2)
Requirement already satisfied: jedi>=0.16 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipython>=5->ipyparallel->ivory==0.0.1) (0.19.2)
[... cut ...] https://github.com/pypa/pip/issues/11457
  Running setup.py develop for museek
Successfully installed museek
(museek) (base) scunnington@compute-201:~/museek$ cd ..
(museek) (base) scunnington@compute-201:~$ python
Python 3.10.4 (main, Jan  3 2024, 14:47:35) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import matplotlib
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'matplotlib'
```

wkhu-astro, 2025-03-18:
I see "Requirement already satisfied: matplotlib-inline>=0.1 in /users/scunnington/environment/museek/lib/python3.10/site-packages (from ipykernel>=6.9.1->ipyparallel->ivory==0.0.1) (0.1.7)" .   There should be available matplotlib. What is the output when you try "which python" ?

stevecunnington, 2025-03-18:
```
(museek) (base) scunnington@compute-201:~$ which python
/users/scunnington/environment/museek/bin/python

```

stevecunnington, 2025-03-18:
Thanks @katrinealice - glad it's not just me! Yeah I'm sure it's a simple fix, but it would be good to fix it in the installation steps...

katrinealice, 2025-03-18:
Yeah I agree. Another person has this issue too, so it's definitely a general thing I'd say

wkhu-astro, 2025-03-18:
A simple method is to install the missing module like 
```
source /users/yourname/environment/museek/bin/activate
pip install 'missing module'
deactivate
```

I have updated the requirements.txt under Museek directory, you can install the required modules using : 
`pip install -r museek/requirements.txt`.    This information is added to the Mussek GitHub document. 

Note: please follow the instructions in the  gain-calibration branch in GitHub, now this branch is not merged to the main.

stevecunnington, 2025-03-19:
Thanks @wkhu-astro - this is working now. **_However_**, it's only corrected on the gain-calibration branch. The error still exists on main

piyanatk, 2025-03-19:
The `requirements.txt` file has also been updated. See https://github.com/meerklass/museek/pull/155

commit 5b70288b (2025-03-19): Merge pull request #155 from piyanatk/fix-matplotlib-import

Update requirements.txt to fix #154